wasp-os/wasp-os #410: make sim fails to find python files

Christian-Prather, 2023-03-19:
## Description of Bug
Hello, firstly thank you for such a cool project!
 
I have spent the weekend trying to get the wasp sim running on my Ubuntu 20.04 system and while a watch with a black screen does display there is the same error no matter what I try. I would love any assistance with this as I really love the wasp-os software but must admit am very frustrated at the moment. I have looked on the issues page and could not see any similar to what I am seeing. Thank you in advance for any help!

## Categroy
Bug / Documentation

## Host OS:
Ubuntu 20.04

## Error:
```
PYTHONDONTWRITEBYTECODE=1 PYTHONPATH=.:wasp/boards/simulator:wasp \
python3 -i wasp/boards/simulator/main.py
Traceback (most recent call last):
  File "wasp/boards/simulator/main.py", line 4, in <module>
    import wasp
  File "/home/christian/Documents/wasp-os/wasp/wasp.py", line 22, in <module>
    import watch
  File "/home/christian/Documents/wasp-os/wasp/boards/simulator/watch.py", line 24, in <module>
    from drivers.cst816s import CST816S
ModuleNotFoundError: No module named 'drivers.cst816s'
```
## Steps to reporduce:
From the wasp-os dir run `make sim`

GaryM0101, 2023-03-19:
Christian, did you follow the installation guide? After doing the "git clone" did you do a "make submodules", 'make softdevice"?
https://wasp-os.readthedocs.io/en/latest/install.html 

Gary

Christian-Prather, 2023-03-19:
@GaryM0101 Thank you for your quick response. Yes I did run both of these commands prior. Based on your comment I decided to delete my repo and attempt from scratch here are the exact order of my steps and the error still occurs

1. `git clone https://github.com/wasp-os/wasp-os`
2. `cd wasp-os`
3. `export PATH=$PATH:/opt/wasp/arm-gnu-toolchain-12.2.rel1-x86_64-arm-none-eabi/bin`
4. `make submodules`
5. `make softdevice`
6. `make sim`

The same error occurs
`ModuleNotFoundError: No module named 'drivers.cst816s'`

Note: the export of PATH is to add my arm gcc toolchain

GaryM0101, 2023-03-19:
I'm sorry to hear that, don't know what else to try assuming you installed the prerequisites prior to doing the git clone.

Christian-Prather, 2023-03-19:
This seems like it is not actually installing the python files as modules? Looking at the make file I don't see how these are ever working to find the drivers folder?

fgaz, 2023-03-19:
Maybe there was an error in one of those steps. Do all of those commands exit with code 0?

closed 2026-11-14 by mbeijen