wasp-os/wasp-os #412: Wasp-os channel added to the official MicroPython discord community server

piersstorey, 2023-03-26:
Hi all,

After some really good dialogue with [Matt Trentini](https://twitter.com/matt_trentini) (Discord profile MattyT#5848) who is one of the core developers of MicroPython, he has kindly created a wasp-os channel in the official discord community server.

@daniel-thompson I hope you don't mind me reaching out to Matt. Unfortunately, I could not find a way of contacting you to check prior to creating the channel. You can contact me on discord Piers#8101 if you'd like to discuss further.

Matt and I hope to promote wasp-os to the greater MicroPython community and to create a space to be available for members of both communities to discuss and support dialogue and questions around the ecosystem. One other consideration would be to reduce the number of issues which could be directed to the channel.

If you would like to join the channel please go to https://discord.gg/9FkEyMezbk 

I'd like to clarify this channel has been created with the absolute best intentions of supporting wasp-os and I have indicated in the channel that I have no direct links with the core team.

Lastly, there has been some really interesting dialogue around porting wasp-os to other watch-like devices including the rp2040 and ESP32 which could further promote the development environment beyond the nRF52 family of microcontrollers.

Any questions, or considerations, please do let me know..

mattytrentini, 2023-05-06:
Just a quick note to say that we've just changed the channel name from `wasp-os` to 'wearables' to cover anything else in the intersection of MicroPython and wearables!

fgaz, 2023-05-06:
The official wasp-os chat is on Matrix and IRC (see https://github.com/wasp-os/wasp-os#community ), maybe your discord channel could be bridged to those

jonnor, 2024-12-27:
Nice to have the chats mentioned. This issue can probably be closed though :) Or at minimum be moved to discussion

piersstorey, 2024-12-27:
> Nice to have the chats mentioned. This issue can probably be closed though :) Or at minimum be moved to discussion

Agreed. We were just reaching out and finding ways of engaging the wider Micropython community in this project 👍

closed 2024-12-27 by fgaz

daniel-thompson, 2024-12-27:
As you can see but the slightly delayed response I was never terribly good at keeping up with the Issues (to be honest I turn them off in most of my projects).

However it is worth mentioning that wasp-os really needs some new people working on it, and one of the major gaps is people who know micro python on the implementation side. As such if anyone in the wearables groups mentions that wasp-os is running an outdated version of uPy then let them know that the deficiency represents a great opportunity to get involved in wasp-os.

piersstorey, 2024-12-27:
> As you can see by the slightly delayed response I was never terribly good at keeping up with the Issues (to be honest I turn them off in most of my projects).
> 
> However it is worth mentioning that wasp-os really needs some new people working on it, and one of the major gaps is people who know micro python on the implementation side. As such if anyone in the wearables groups mentions that wasp-os is running an outdated version of uPy then let them know that the deficiency represents a great opportunity to get involved in wasp-os.

Thanks for your response @daniel-thompson, I would like to see this project grow and agree, for that to happen it needs new people getting involved which is one of the reasons for getting exposure to wearables in the main MicroPython community Discord channel. I know in speaking with @mattytrentini there is an opportunity for positive dialogue between the MicroPython ecosystem and Wasp-os and would encourage a discussion to understand what can be done to support the project and promote collaboration.