wasp-os/wasp-os #417: "make check" returns pytest Error 127

thiswillbeyourgithub, 2023-04-21:
Hi!

I'm not familiar with pytest and can't seem to make `make check` command in the master branch.

I get instantly : 
```
PYTHONDONTWRITEBYTECODE=1 PYTHONPATH=.:wasp/boards/simulator:wasp:wasp/apps/system \
pytest-3 -v -W ignore  wasp/boards/simulator
/bin/sh: 1: pytest-3: not found
make: *** [Makefile:126: check] Error 127
```

Can anyone give me pointers on this?

fgaz, 2023-04-21:
> pytest-3: not found

If your `pytest` isn't called `pytest-3` (mine also isn't) you can override it with `PYTEST=name_of_pytest`. For example mine is called just `pytest` so I do `make check PYTEST=pytest`.

fgaz, 2023-04-21:
#419 could help avoid this kind of issue

PR #419 (2023-06-21): Use `python -m` to run pytest
Not every distro calls it `pytest-3`.
By using `$(PYTHON) -m` we ensure it's coherent with the python install

Closes #417
closed 2023-04-21 by thiswillbeyourgithub