wasp-os/wasp-os #423: Stuck in Infinitime bootloader

agittins, 2023-05-23:
On my devwatch I was running infinitime 1.11. I fired up nrfconnect (which I think had updated to a recent-ish version), and went through the dfu steps. The watch showed the infinitime "firmware update" screen with progress bar. I don't recall the progress bar moving at all, but nrfconnect showed a bit of data transferred (maybe 10% of the graph) and then the watch rebooted.
It booted back up into infinitime. So I long-pressed the button, and now the watch only boots into the plain white "Infinitime" screen, with the two triangles and "infinitime" (this is after going through the infinitime v1.0.0 bootloader with the big white pinecone, filling green then (if I hold the button) red and blue.

If I release the button in the "blue" stage, the watch reboots and goes through the big white v1.0.0 pinecone again. If I let it fill green, it goes there as soon as it fills up, or if I hold it through red it takes a bit longer at the top then reboots, fills green and goes to the same white triangle logo screen.

While in the white logo screen I can see the watch in nrfconnect. Hitting connect spins for a while and fails with no error. At no time does the (dfu) button appear in the top-right.

I also tried the ota-dfu from the repo. With the legacy flag I get:
```
$ ./tools/ota-dfu/dfu.py -z build-pinetime/reloader-mcuboot.zip -a $DEVWATCH --legacy
Sending file reloader-mcuboot.bin to C7:B8:C6:B0:27:11
Binary imge size: 303764
Binary CRC32: 2117925679
Connecting to C7:B8:C6:B0:27:11
Checking DFU State...
char-read-uuid 00001534-1212-efde-1523-785feabcd123
Need to switch to DFU mode
Exception at line 154: UUID not found: 00001531-1212-efde-1523-785feabcd123
```

and without, I get:
```
$ ./tools/ota-dfu/dfu.py -z build-pinetime/reloader-mcuboot.zip -a $DEVWATCH
Sending file reloader-mcuboot.bin to C7:B8:C6:B0:27:11
Binary imge size: 303764
Binary CRC32: 2117925679
Connecting to C7:B8:C6:B0:27:11
Checking DFU State...
Exception at line 169: UUID not found: 8ec90001-f315-4f60-9fb8-838830daea50
```

I've also tried downgrading nrfconnect to 4.24 and I still can't connect, let alone get the option to dfu.

I remember getting a bit stuck like this last year when I was switching from wasp-os to infinitime but managed to sort it *somehow*, just not sure how!

fgaz, 2023-05-23:
If I remember correctly, there's a nrf connect setting (packet size or something) that makes flashing more reliable

fgaz, 2023-05-23:
Also there's a specific nrf connect version that doesn't work, I think it's mentioned in one of the issues

agittins, 2023-05-23:
Cheers, yes there's PRN and you can change the default of 10 packets to a lower number. I've got it set at 1, but I'm not even getting as far as the dfu button appearing in the nrf ui.

I think the current version (at least at around Oct last year) of nrf wasn't working. I've tried 4.25.0, 4.24.0, 4.24.3 and 4.26.0 with no luck so far (one of those used to work). But since I'm not able to connect at all I'm a bit concerned that the recovery firmware (which I think is what I'm in) isn't working correctly. I might need to head that way for help given it's the codepath I'm in, but thought some here may have run into it.

agittins, 2023-05-23:
Aha! dfu.py from infinitime's repo sent the reloader OK, while nrfconnect and the wasp-os dfu.py would not.

Then it was sitting at the wasp-os pinecone, and the infinitime dfu.py would not work, but the wasp-os one does!