wasp-os/wasp-os #424: Wasp A La carte - Wasptool not accessable using standadard toml out-of-the-box build

GaryM0101, 2023-06-03:
I've installed the al-la-carte version on two Pinetime devices. Both are out-of-the-box installs. I cannot get to the wasptool console for both, timeouts... I also tried safe mode, but I cannot get there either. Bluetoothctl and Gatttool connect successfully on both. Gadgetbridge also connects to both without issue. I have a nRF52 Dev kit with an earlier version of wasp, and wasptool works just fine. I suspect this is a memory issue with the default al-la-carte build. Has anyone else experienced the same? If I have time, I'll strip it down to bare bones and see if that's really the issue.

Running arm7L Debian Linux

beardeddude, 2023-06-03:
I'm running early ALC builds (the first commit after merge) on two pine times and can connect to the wasptool console without issue. Im running linux mint (debian) based system. So it is probably either something about your environment or a change made after ALC was merged.

I wouldnt think ALC is related to this issue as it only controls what apps are included in the build but I guess its not impossible.

GaryM0101, 2023-06-03:
You might be on to something. I flashed 0.4 and couldn't connect. So I did a sudo apt upgrade, and after the reboot, my "Infinitime Pinetime" started flashing a Bluetooth passcode and I was prompted with a window to enter it. I installed ITD not long ago to access the extended memory. It could be that somehow ITD put a secure wrapper on my Bluetooth connection. Though it doesn't explain why the nRF52 connects.

GaryM0101, 2023-06-03:
Well...what a dummy I am. The wasptool had a mac address lock on my nRF52dk and wouldn't let it go even by trying to connect directly to it using --device XX:XX:XX:XX:XX:XX 

Remove the locked mac address by running "bluetoothctl remove XX:XX:XX:XX:XX:XX"

That's what I get for having too many wasp device here on the desk. Sorry @beardeddude for sending you on a wild goose chase. 

Gary

closed 2023-06-03 by GaryM0101