wasp-os/wasp-os #425: Send notification via Bluetooth

jfekete1, 2023-06-03:
Hi all,

I'm trying to send a notification to my watch vie bluetooth using a Linux command line tool like bluetoothctl or gatttool.
So far I had no success.
I used gadgetbridge and sniffed the bluetooth packets sent by the android application, this is what I saw with wireshark:
Service UUID: 6e400001b5a3f393e0a9e50e24dcca9e
write command 0x52
handle 0x0013 (Nordic UART Service: Nordic UART Tx)
value 104742287b227422   ->  GB({"t"

write command 0x52
handle 0x0013 (Nordic UART Service: Nordic UART Tx)
value 3a226e6f74696679   ->  :"notify

write command 0x52
handle 0x0013 (Nordic UART Service: Nordic UART Tx)
value 222c226964223a31   ->  ","id":1

write command 0x52
handle 0x0013 (Nordic UART Service: Nordic UART Tx)
value 3638353737393230   ->  68577920

write command 0x52
handle 0x0013 (Nordic UART Service: Nordic UART Tx)
value 362c22626f647922   ->  6,"body"

write command 0x52
handle 0x0013 (Nordic UART Service: Nordic UART Tx)
value 3a2268616c69222c   ->  :"hali",

write command 0x52
handle 0x0013 (Nordic UART Service: Nordic UART Tx)
value 2273656e64657222   ->  "sender"

write command 0x52
handle 0x0013 (Nordic UART Service: Nordic UART Tx)
value 3a22c96e222c2274   ->  :" n","t

write command 0x52
handle 0x0013 (Nordic UART Service: Nordic UART Tx)
value 656c223a222b3336   ->  el":"+36

write command 0x52
handle 0x0013 (Nordic UART Service: Nordic UART Tx)
value 3730333038393532   ->  70308952

write command 0x52
handle 0x0013 (Nordic UART Service: Nordic UART Tx)
value 33227d290a         ->  4"}) 

So basically it sent this json using 11 write commands:
GB({"t":"notify","id":1685779206,"body":"hali","sender":" n","tel":"+36703089524"})

The id is the current unix timestamp.
However I'm unable to replicate this with gatttool or bluetoothctl.
Can someone provide a python script in tools folder, that can send a notification?

GaryM0101, 2023-06-03:
I'm sure you can craft a script using the PagerApp. Have a look at the pager.py in the wasp/apps/system directory, 

Cheers, Gary

jfekete1, 2023-06-03:
> I'm sure you can craft a script using the PagerApp. Have a look at the pager.py in the wasp/apps/system directory,
> 
> Cheers, Gary

Thank you Gary, I'll definitely try it out!

jfekete1, 2023-06-07:
Okay, I added the notification feature I needed in commit 12dca2b8e986388b4ed38c17f74137f5553bcd27
I also tested it out using it like this:
wasptool --send-notification
wasptool --send-notification --title hello
wasptool --send-notification --title hello --body world
wasptool  --device D3:7D:16:55:E4:f2 --send-notification --title hello --body 'asd asd asd asd'

It works perfectly! :)
Thank you Gary and fgaz !

jfekete1, 2023-06-30:
@fgaz can you check my commit?
f85da08