wasp-os/wasp-os #431: Application Library docs page is empty

fgaz, 2023-06-27:
https://wasp-os.readthedocs.io/en/latest/apps.html is empty

I think this happened with #403 by @beardeddude

beardeddude, 2023-06-27:
This page builds for me locally. So something must be different in the gitlab workflows build. I'm looking into it. Let me know if you are able to reproduce it locally at all

beardeddude, 2023-06-30:
As far as I can tell there is no issue here. The app page builds as expected. In the last successful build of the master pipeline we can see the page correctly generated in the artifact (bottom most archive starting with wasp-os) https://github.com/wasp-os/wasp-os/actions/runs/4992976604 .

The only issue I can find is that the most recent 2 builds in the master pipeline have failed due to networking issues.
https://github.com/wasp-os/wasp-os/actions/runs/4993074589/jobs/8941664723
https://github.com/wasp-os/wasp-os/actions/runs/5337524920/jobs/9673684154

Generally this kind of issue is temporary. I'm betting if someone with the needed permissions were to re-run the master pipeline this issue would fix itself.

fgaz, 2023-06-30:
Makes sense, thanks for investigating. Let's keep this open until @daniel-thompson or the next commit to master triggers a rebuild

fgaz, 2023-09-27:
Upon further investigation, it looks like this happens because readthedocs does not run `make docs`, but runs sphinx directly, so it can't find appregistry.

I'm trying to fix it in #463

PR #463 (2023-09-27): Add .readthedocs.yaml
It is now mandatory
https://blog.readthedocs.com/migrate-configuration-v2/

Also fixes #431 (empty app library in rtd)
closed 2023-09-27 by daniel-thompson