wasp-os/wasp-os #436: Fails to compile - micropython folder unexpectedly empty

tomchiverton, 2023-07-12:
Ubuntu 22.04 LTS 

This is a fresh clone from master, and I've followed https://wasp-os.readthedocs.io/en/latest/install.html#building-wasp-os-from-source

I had to use `pip3 install --user --break-system-packages cbor pysdl2 tomli` as the final step because pip moaned.

```
falken@wopr:~/workspace/wasp-os$ make -j `nproc` BOARD=p8 all
mkdir -p build-p8
(cd wasp; ../tools/preprocess.py boards/p8/watch.py.in > boards/p8/watch.py) \
        || (rm -f wasp/boards/p8/watch.py; false)
rm -f bootloader/_build-p8_nrf52832//p8_nrf52832_bootloader-*-nosd.hex
make -C bootloader/ BOARD=p8_nrf52832 all genhex
make[1]: Entering directory '/home/falken/workspace/wasp-os/bootloader'
make[1]: *** No rule to make target 'all'. Stop.
make[1]: Leaving directory '/home/falken/workspace/wasp-os/bootloader'
make: *** [Makefile:38: bootloader] Error 2
make: *** Waiting for unfinished jobs....
```
Based on that I tried with no build target, still no joy
```
falken@wopr:~/workspace/wasp-os$ make -j `nproc` BOARD=p8 
rm -f bootloader/_build-p8_nrf52832//p8_nrf52832_bootloader-*-nosd.hex
make -C micropython/mpy-cross
make -C bootloader/ BOARD=p8_nrf52832 all genhex
make[1]: *** micropython/mpy-cross: No such file or directory. Stop.
make: *** [Makefile:62: micropython] Error 2
make: *** Waiting for unfinished jobs....
make[1]: Entering directory '/home/falken/workspace/wasp-os/bootloader'
make[1]: *** No rule to make target 'all'. Stop.
make[1]: Leaving directory '/home/falken/workspace/wasp-os/bootloader'
make: *** [Makefile:38: bootloader] Error 2
falken@wopr:~/workspace/wasp-os$ 
```

`git submodule init && git submodule update` needs adding to the docs somewhere ?

fgaz, 2023-07-13:
> I had to use pip3 install --user --break-system-packages cbor pysdl2 tomli as the final step because pip moaned.

That's weird, `--user` should be enough and `--break-system-packages` should never be necessary. 

> git submodule init && git submodule update needs adding to the docs somewhere ?

yes, `make submodules` is missing from "install prerequisites"

tomchiverton, 2023-07-29:
Docs have been updated

closed 2023-07-29 by tomchiverton