wasp-os/wasp-os #437: Compile fails -  nrfx.h: No such file or directory

tomchiverton, 2023-07-12:
Ubuntu 22.04 LTS

Fresh clone of master, I've init'd and update'd the modules, and run `./micropython/ports/nrf/drivers/bluetooth/download_ble_stack.sh` as prompted.

I've set the PATH to the ARM toolkit by `export PATH="$PATH:/home/falken/workspace/gcc-arm-none-eabi-10-2020-q4-major/bin/"
`

Build still fails due to some missing header files.

```
$ make -j `nproc` BOARD=p8 
rm -f bootloader/_build-p8_nrf52832//p8_nrf52832_bootloader-*-nosd.hex
make -C micropython/mpy-cross
make[1]: Entering directory '/home/falken/workspace/wasp-os/micropython/mpy-cross'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
make -C bootloader/ BOARD=p8_nrf52832 all genhex
make[1]: Entering directory '/home/falken/workspace/wasp-os/bootloader'
CC main.c
src/main.c:41:10: fatal error: nrfx.h: No such file or directory
   41 | #include "nrfx.h"
      |          ^~~~~~~~
compilation terminated.
make[1]: Leaving directory '/home/falken/workspace/wasp-os/micropython/mpy-cross'
rm -f micropython/ports/nrf/build-p8-s132/frozen_content.c
make -C micropython/ports/nrf \
        BOARD=p8 SD=s132 \
        MICROPY_VFS_LFS2=1 \
        FROZEN_MANIFEST=/home/falken/workspace/wasp-os/wasp/boards/p8/manifest.py \
        USER_C_MODULES=/home/falken/workspace/wasp-os/wasp/modules
make[1]: Entering directory '/home/falken/workspace/wasp-os/micropython/ports/nrf'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from /home/falken/workspace/wasp-os/wasp/modules/bma42x-upy
make[1]: *** [Makefile:368: _build-p8_nrf52832/main.o] Error 1
make[1]: Leaving directory '/home/falken/workspace/wasp-os/bootloader'
make: *** [Makefile:38: bootloader] Error 2
make: *** Waiting for unfinished jobs....
make[1]: *** No rule to make target 'bma42x-upy/BMA423-Sensor-API/bma4.c', needed by 'build-p8-s132/genhdr/qstr.i.last'. Stop.
make[1]: Leaving directory '/home/falken/workspace/wasp-os/micropython/ports/nrf'
make: *** [Makefile:64: micropython] Error 2
```

tomchiverton, 2023-07-12:
I'd hoped that wasptool could do this, but sadly `$ ./tools/wasptool --exec /home/falken/workspace/wasp-os-Elenedeath/wasp/drivers/st7789.py --eval "watch.touch = ST7789(watch.i2c)"`

```
[##################################################] 100% 
~~~
MemoryError: memory allocation failed, allocating 501 bytes
Traceback (most recent call last):
  File "<stdin>"
SyntaxError: invalid syntax
~~~
~~~
Traceback (most recent call last):
  File "<stdin>", in <module>
TypeError: function takes 3 positional arguments but 2 were given
~~~
```

`./tools/wasptool --eval "watch.display.invert(False)"` takes care of the first change required 

`./tools/wasptool --eval "watch.display.write_cmd( const(0x36) ); watch.display.write_data(b'\x40') "` takes care of the rest.

This is non permanent and will be lost at reboot I think

fgaz, 2023-07-13:
Are you sure the modules are properly initialized? What happens if you do `make submodules` first?

tomchiverton, 2023-07-15:
I'd missed that step, thanks !

closed 2023-07-15 by tomchiverton