wasp-os/wasp-os #438: I think I accidentally bricked my pinetime

Onni3000, 2023-07-15:
Today I tried installing wasp os to my pinetime. I was able to install the bootloader without problems, but now when I try to install the wasp os to it, its just stuck at 0%

(For some reason it didn't allow me to put the pictures here so Ill just put them into the comments.)

Onni3000, 2023-07-15:
[image]

[image]

SteveAmor, 2023-07-15:
Did you use nRF Connect to install the bootloader? Recent versions of nRF Connect have stopped working for installing firmware.  Can you use another program like Gadgetbridge?

Onni3000, 2023-07-15:
> Did you use nRF Connect to install the bootloader? Recent versions of nRF Connect have stopped working for installing firmware. Can you use another program like Gadgetbridge?

Thank you for response!

I first tried to install with nRF Connect, but it didn't work so I decided to try to do it with Gadgetbridge (which worked), but Gadgetbridge apparently doesn't support this device now that it has this bootloader so I am unable to use it to install the micropython.zip.

Onni3000, 2023-07-15:
I was able to get Wasp OS installed after downloading an older version of nRF Connect.

closed 2023-07-15 by Onni3000

Brulijam, 2024-02-26:
Hi I have the same problem. Which version of nrf connect worked for you?

GaryM0101, 2024-02-27:
You can try 4.24.3, versions after that are no longer supported.

 

Links:
------
[1]
https://github.com/wasp-os/wasp-os/issues/438#issuecomment-1965532558
[2]
https://github.com/