wasp-os/wasp-os #445: more deepsleep ?

thiswillbeyourgithub, 2023-09-02:
Hi,

Not very knowledgeable about any of this so I preferred to ask. Recently Infinitie managed to get more than 10 day of battery life which is dramatically longer than what we can currently get out of the waspos.

My question is : is there a way to modify the balance between usability and battery? I would be very interested in a less responsive watch that spends more time in deep sleep for example.

Additionnaly, is anyone interested by listing the different milestones needed to improve battery life? I think I remember something about the bluetooth beacon being always on for example?

Cheers

GaryM0101, 2023-09-08:
Infinitime released a battery fix in their latest release. Just tested
it on the Pinetime and got 28 days till the battery got to 18%.
Impressive! 

 

Links:
------
[1] https://github.com/wasp-os/wasp-os/issues/445
[2]
https://github.com/

thiswillbeyourgithub, 2023-09-18:
I addde a draft pr at #453 for anyone interested in trying to make the watch less responsive but lasting longer.