wasp-os/wasp-os #449: I can't scroll down in the software app

EmmanuelGoldstein, 2023-09-07:
I can't scroll down in the software app on my PineTime. When I have the official release of Wasp-os installed it works fine, but when I use any of the newer CS versions, it won't let me scroll to the next page in the software app, so I can't activate most of the apps. It only affects the software app (I can scroll fine through other apps) and it also only affects the newer CS releases, it works fine on the old v0.4.1 official release.

fgaz, 2023-09-26:
By CS you mean CI? Do you have any files/directories in the flash memory? Are there arrows in the bottom right of the screen like in the following image?

[image]

EmmanuelGoldstein, 2023-09-28:
I installed the official 4.1 release after the others didn't work. But if I remember correctly it did show the button on the lower right, it just wouldn't scroll when you pushed it or swiped the screen. Other apps worked fine as long as they were activated by default. I had no additional files/directories.

dekuraan, 2024-01-23:
Hi, sorry to comment so late, but I was wondering if you ever found a fix to this, I have the same issue, I have just been using the CI builds, do you know if I should flash micropything from wasp-os-commithash or pinetime-commithash? or does it not matter

dekuraan, 2024-01-23:
I was under the impression that all software came installed by default, but I just looked at wasp.toml :facepalm:

fgaz, 2024-01-24:
@dekuraan There isn't enough space on the device for all the apps unfortunately. You'll have to build wasp-os yourself if you want to choose which apps are included. Or you can upload the .mpy apps from the `apps` tarball from CI, but those take more memory to run.

I'm planning to set up a website where you can request custom wasp-os builds, so building the os yourself won't be required anymore, but I've had other priorities recently...

fgaz, 2024-01-24:
Closing this, since OP likely had the same issue, but feel free to continue the discussion

closed 2024-01-24 by fgaz