wasp-os/wasp-os #458: /flash - how should we lay out the directory structure?

drtonyr, 2023-09-22:
We are often short on space, but in fact we have oodles of space in /flash (4MB SPI NOR XT25F32B), just it's not commonly used.

At the moment we have main.py and a logs directory.   I'd like to see a policy that gives us a clean filing system, which means that apps have their own space.

I have a strong preference for not using plurals in names (e.g. file names or variable names), but WASP-OS doesn't, so to keep things simple I'll only suggest plurals.

That said, here are a few ideas.  Firstly we need a naming convention for apps, here are some choices following [app-naming-conventions-and-placement](https://wasp-os.readthedocs.io/en/latest/appguide.html#app-naming-conventions-and-placement)

1. the file name in snake_case
2. the class name in pascal case
3. the class name with App appended
4. the value of the NAME variable

I'm in favour of (2), it's the one that is used in most places.  Whatever is chosen, lets call it $APP.   Now we have some choices as to how we lay out the file system:

1. top level - /flash/$APP.$EXT for one or more values of $EXT, e.g. log, cfg, etc.
2. distributed according to type, /flash/$EXT/$APP.*.   This is what we have now with /flash/logs, we could also have /flash/etc or /flash/cfg
3. distributed according to APP with no top level directory, /flash/$APP/* - developers can put what they like under  /flash/$APP/
4. distributed according to APP with a top level directory /flash/apps/$APP.$EXT or /flash/apps/$APP/*.   If only one file is needed then developers can use /flash/apps/$APP.$EXT, if more than one then they must create a directory
5. distributed according to APP with an enforced directory /flash/apps/$APP/*

I'm in favour of (5), it has the advantage that within the directory fixed names could be used, like /flash/apps/$APP/ICON.rle and the developer is pretty much free to do what they want in this space, which is important as all apps are different.  Flash space is accountable for (not the case if files are distributed) and can be easily cleaned up.  I'd also be happy with (3).

To give some concrete examples, I'm currently working on TideClock that needs a text location database.   So I could have /flash/apps/TideClock/locations.txt where users could enable (uncomment) one or a couple of favourite locations and the clock face could scroll through these.   I've also go a 'working' editor, if people want it then I'll need some default space to create new files and I'd do that with /flash/apps/EditExec/$DATE.txt.   I also like the idea of reading the icons from /flash, but I haven't done any timings.

What do people think?   I'm hacking right now, so thinking about how my app can be configured.  There's no great rush, but it would be good to make an informed choice.

beardeddude, 2023-09-23:
Like you im in favor of 2 and 5 basic patterns. But with one small tweek. The app dir in flash is used to load programs and files in that dir will become available in the software app when running on the device (there are some big memory limitations running a program this way but it is functional). The app dir in the root of the project is the same for the sim. Instead of changing where the root of the project is and having different mappings and sync I think a lighter touch would be to just create an appdata dir in the root of the project and to have an appdata dir in the root of the device flash.

/appdata/$APP/*

The wasptool currently supports pushing and pulling files from flash on the watch. Not all apps in the project are loaded to the device (you configure via wasp.toml). Its probably enough to just put a note in the readme of your app that the user should push the appdata folder to their device.

thats my 2 cents for what its worth.

daniel-thompson, 2023-09-24:
Don't use absolute paths for this. It is a waste of flash (and makes things harder on the simulator). On the watch the working directory should be /flash and for the simulator it is the top-level of the sources.

Currently the haiku app follows #1 above.

drtonyr, 2023-09-24:
@beardeddude @daniel-thompson - thanks, I've now understood that we already have some mapping of the file space, specifically /flash on the watch ends up as the top of the source tree.   This makes 'apps' a terrible choice, as it's already in use for source code.   It also means that we need some directory name else the source code would get polluted.

Yes, the Haiku app uses /flash, I'm pleased to see that it can read haiku.rle if present.  The fact that neither haiku.rle nor haiku.txt are in the source, and if they were they'd be polluting the top level, suggests that something needs to be done.

@beardeddude, did you mean /appdata/$APP/* or /flash/appdata/$APP/*?   I can see creating a directory in the source called appdata and then creating appdata/haiku/haiku.{rle,txt} which appears on the watch as /flash/appdata/haiku/haiku.{rle,txt}.

Fundamentally there's a conflict, the simulator likes /flash/apps to read app source code from, but for that to be clean in the source tree we are going to have to have a fixed prefix for every file that will take up space.  Maybe a new directory called 'data' at the top level of the source that gets mirrored to the watch, with no enforces subidirs?  Then data/haiku.{rle,txt} could be distributed and work and it would only be another 5 bytes.   (and data is a plural <ducks>)

fgaz, 2023-09-26:
My preferences are:

* (1) or (2) for the name, since (3) is just redundant and (4) is not guaranteed to be unique
* (4) or (5) for the structure, with a directory named `data`. (1) is a mess. (2) is nice on servers and desktops but it doesn't make much sense on wasp os. (3) has the problems described above

drtonyr, 2023-09-28:
> @drtonyr Without using absolute paths it would be "appdata/APPNAME/data.txt". If we had the "appdata" dir in the root of the project that could store the files for each the user could just push the whole folder to their watch or just the apps they wanted and it would be the same in the sim and the watch.
> 
> Ie: in the haiku app it would be
> 
> `open('appdata/haiku/haiku.txt')`

Yup, I agree.   This gives us one directory which is under git which can be synced to the watch for all the apps to use.  This is pretty powerful, it keeps things clean, it allows us to offload data from the program storage and it's easy to understand and work with from a developer point of view.

I also take Daniel's point that long names take up storage.

Combining the two, I'm using open('data/TideData.txt') for now.   I don't know when my app will be finished as I'm currently distracted by getting text cleanly written to the display.  When finished I'll reference this thread in my PR and that might be a good time to make a decision.

drtonyr, 2023-10-08:
#470  is another app that would like to use SPI /flash.

Eloitor, 2023-10-08:
My preference: (5) > (3)  > (4)  > (2) > (1).

What about 

`flash/cache/$APP/*` for cache files that are not important and can be deleted (maybe for some app this could be useful?)
`flash/apps/$APP/*`  for files specific to apps, configs, data, ...

kj7rrv, 2023-10-09:
Option 5 looks best to me. For the Bible app, I would use `/flash/apps/bible/KJV/` for the text, at least in the current KJV-only version. If/when support for multiple translations is added, other Bibles could go under e.g. `/flash/apps/bible/RVG`. (Two complete Bibles won't be possible due to storage limitations, as a single copy of the KJV nearly fills the flash memory, but partial copies are feasible; two different NT translations should fit.)

kj7rrv, 2023-10-09:
As far as the choice of name, I would suggest that the directory name should exactly match the app's Python file name, minus `.py`, so that would mean `snake_case`. Any of the options would be fine, though.

daniel-thompson, 2023-10-09:
Calling the data directory apps/ won't play nice with the simulator (since we can't add the top-level apps/ directory to .gitignore since it's already being used). data/ is likely a better choice.

drtonyr, 2023-10-09:
> Calling the data directory apps/ won't play nice with the simulator (since we can't add the top-level apps/ directory to .gitignore since it's already being used). data/ is likely a better choice.

Agreed that 'apps' doesn't work for us as the simulator uses it.

I like 'data' - to me it says that this is where data is stored, something that we need to sync to the watch.

I'd like to explore your .gitignore comment.   I was viewing this directory as something that exists in git, but that's because I have a relatively small data file.  I don't want to embed this in the code as I want the user to edit it (set a location by uncommenting a line).  If it's in git then I could push my app and it's data and if accepted then it would run for everyone.   However, with the bible app under development then this doesn't work - I don't want my SPI flash taking ages to sync if I'm not going to use the app.

I don't have a solution, I've just realised that I have one more issue to think about.

kj7rrv, 2023-10-09:
Could apps opt in by setting an attribute on the class (e.g. `self.isolate_storage = True` in `__init__()`)?

drtonyr, 2023-10-16:
> Could apps opt in by setting an attribute on the class (e.g. `self.isolate_storage = True` in `__init__()`)?

If an app needs to read data under /flash we have to get that data there somehow.  If we've pushed it there, then we have an app that needs data, so we can os.chdir() there.   So maybe we don't need more code in __init__() at all, just `try os.chdir()/except` or `os.path.exists() and os.chdir()`  (I'm a little afraid of exceptions in micropython, seems we don't have the memory).

This has a couple of advantages:
- no change to existing apps, they can upgrade if and when they want to
- the data directory name is mostly hidden, so it should be less contentious to decide on the name

We still need to think about what gets synced and what doesn't and how the data gets synced, but that can evolve later.

daniel-thompson, 2023-10-16:
Automatic opt-in on directory existence sounds like a very neat trick. I like that. It should work flawlessly with existing apps that already require manual population (such as the haiku app). It is also good for avoiding flash directory layout regressions when upgrading (since old layouts also work seamlessly after the transition).

Note that some apps log data rather than read it and don't require pre-population of the directory structure. The step tracker is an example of such an app. That could easily be solved with an (optional) .init_storage() API though. It can create the directory based on the class name (if it does not exist) and change into it.