wasp-os/wasp-os #462: Create default main.py if missing, not only when formatting

fgaz, 2023-09-27:
Recent versions of infinitime make use of the flash memory and will format it, so it can happen that wasp-os does not create a default main.py on first boot.

I propose to always create a default main.py if the file is missing, or alternatively to behave as if it was there

GaryM0101, 2023-11-06:
This happened today when Beatlink did a clone and new build of wasp. I'd go with your recommendation above. 

_the [main.py](http://main.py/) upload did the trick
not sure why the flash didnt write a new one_

GaryM0101, 2025-03-22:
We had another example of this a couple of days ago. I don't think it hurts to create a default main.py if one is not found. I'll test one out and open a PR.

[M] <J​ohn Player> Decided to give WaspOS a try and clicking the find my watch button in GB killed it. Now it just boots to a screen that says "main.py"  Bootloader seems fine so I think I'll be putting infinitime back on later today.
 — 
3/20/25, 7:52 AM
[M] <J​ohn Player> And now I'm bootloop locked on a version of the watch that's supposedly safe to switch back from. Amazing.
[M] <J​ohn Player> Python ruins everything