wasp-os/wasp-os #472: nRF Connect 4.26.0 (latest version) fails to install the Wasp boot loader - Pinetime

GaryM0101, 2023-10-18:
The latest version of nRF Connect 4.26.0 fails to install the Wasp boot loader on the Pinetime.  The workaround is to have the user install an older version, in this case 4.24.3.  The Infinitime project now uses Gagetbridge for their installs; they found every new nRF version was failing to do updates for them. 

We should document that folks use the older version of nRF Connect 4.24.3 for installing Wasp as the workaround for Andriod.
https://github.com/NordicSemiconductor/Android-nRF-Connect/releases

Are there any other DFU phone apps like nRF Connect that can use for the Pinetime, Gadgetbridge maybe?

GaryM0101, 2023-10-18:
Well...it has been confirmed by a user that gadgetbridge can indeed install Wasp and then go back to Infinitime. We'll have to update the docs for this as well.

fgaz, 2023-10-18:
Yes I remember using gadgetbridge too and it worked well

GaryM0101, 2023-10-18:
We might add this troubleshooting step as well. 

_Caleb Fontenot (He/Him) -- Today at 4:42 AM_
_Yes, but as a troubleshooting step, I had uninstalled GadgetBridge when
I was trying to flash the watch and GadgetBridge only lets you flash
devices you've already paired with it first. Because the PineTime was
stuck in DFU mode, it wouldn't pair with my phone, so I couldn't use
GadgetBridge..._

 

Links:
------
[1]
https://github.com/wasp-os/wasp-os/issues/472#issuecomment-1768545991
[2]
https://github.com/

GaryM0101, 2024-08-07:
Close with implementation of PR #480

closed 2024-08-07 by GaryM0101