wasp-os/wasp-os #474: how to use bluetooth

JustAnotherGameDevWasTaken, 2023-10-26:
i don't seem to have ubluepy available in the sim, it says module not found. importing ble works, but i just have "enable()", "disable()", and "enabled()". i'm not sure what i'm doing wrong.

fgaz, 2023-10-26:
What do you want to do specifically? If you want to communicate with gadgetbridge or similar, there's gadgetbridge.py

JustAnotherGameDevWasTaken, 2023-10-26:
i'm trying to familiarize myself with the tools and modules available via the sim while waiting for a pinetime watch i ordered to arrive.

i'm wondering why i don't have ubluepy available as a module. i'm assuming that the ble module only has the three commands listed above since it's the sim and not the actual watch.

i'm mainly looking to create projects that involve sending and receiving data between python scripts over bluetooth using a pinetime as the interface.

GaryM0101, 2023-10-26:
My understanding is the simulator only uses functions as defined in the ble.py in the simulator directory which is the ble module. Python3 is called for the simulator which has no ubluepy module by default.

beardeddude, 2023-10-26:
From the python side of things you only get the 3 functions. You can send data from and to the watch using the serial interface. Basically printing out prints data out over the connection. If you want control over more control over the ble radio that would need to be done in the C code.

Look at how the following is working for examples:
music_player.py
phone_finder.py
pager.py
gadgetbridge.py
wasptool.py

JustAnotherGameDevWasTaken, 2023-10-26:
I'm looking at wasptool and I'm not sure I'm understanding how it works.. what I'm looking to do is send data to and from the watch to a python script running on a computer.
I'm quite new to python so I'm not sure how to do that currently. Doing it over serial seems easier than messing with Bluetooth apis directly, but I'm not sure how the scripts work in terms of opening a Bluetooth serial connection.

GaryM0101, 2023-10-26:
Take a look at the [Application Writer's Guide](https://wasp-os.readthedocs.io/en/latest/appguide.html). This a good place to start for us beginners in python and using the wasptool.

GaryM0101, 2023-10-26:
Here is the way I connect my Pinetime using bluetoothclt:

$ bluetoothctl scan on
Discovery started
[CHG] Controller DC:A6:32:14:2E:68 Discovering: yes
[NEW] Device FA:38:CE:C5:3C:ED PineTime

$ bluetoothctl connect FA:38:CE:C5:3C:ED
Attempting to connect to FA:38:CE:C5:3C:ED
[CHG] Device FA:38:CE:C5:3C:ED Connected: yes

Once connected, use the wasptool script as indicted in the guide

JustAnotherGameDevWasTaken, 2023-10-27:
Had another read through of wasptool and I have a vague idea of how to handle the connection on the computer. Still unsure of how to do serial stuff on the watch side. I assume sending data from the watch can just be a print statement? Closing this issue because I feel like it's got off topic.

closed 2023-10-27 by JustAnotherGameDevWasTaken