wasp-os/wasp-os #481: Slow display of picture from user storage

Tuxicoman2, 2023-11-05:
Hi,

I got the watch a week ago. Already tried to improve the software part :)

I did a watchface that uses a 240x240px picture as background. The RLE image  (113ko BMP file) is located in the user storage and uses the method specified in the gallery app to load & push the data to the screen driver line by line in order to be RAM efficient.

However, the process is very slow. The watchface takes 5sec to be displayed (switch on & off the screen is fast, using the siumlator is fast)

What do you think the bottleneck problem is? storage read access time ? or is there a need for a better code to display larger buffer?

fgaz, 2023-11-05:
Decoding BMP is a bit slow, you could use raw rgb565 data. You can see how it was done in the gallery app before 61d9dbce7c7656a0411ce2b9ee8f4ca84643f47b:

https://github.com/wasp-os/wasp-os/blob/eba94cd4f63f17f292507fbde1ed743c9f353449/apps/Gallery.py

If the image compresses well and you ind that the bottleneck is storage speed you can also try to RLE-encode it like icons

daniel-thompson, 2023-11-05:
Looking at the current Gallery code I suspect that the endianess swap between the load from filesystems and write to the display is likely to be a problem!

The hardware should be capable of taking an uncompressed image from SPINOR and writing it to the display in about a quarter of a second. There are a few overheads that means getting down to raw hardware performance would be very difficult but 5 seconds is way too much.

It is the work in the inner loops that matters most and all I can really see there is filesystem overhead (which would be hard to avoid) and the byte-swap loop (which can be avoided by byte-swapping on the PC before copying the image to the user storage).

PS You could also use [viper code emitter](https://docs.micropython.org/en/v1.9.3/pyboard/reference/speed_python.html#the-viper-code-emitter) for the code that does the byte-swap. Viper code is a lot faster than regular python code (but still not great) and we use it to optimize several key wasp-os drawing functions. However the fastest thing to do is not need to byte-swap!

Tuxicoman2, 2023-11-13:
> Decoding BMP is a bit slow, you could use raw rgb565 data. You can see how it was done in the gallery app before [...]

So I implemented that. That is a bit faster indeed.
Strangely, there is a misalignment of the picture at 2 rows on the device (the simulation runs fine but I don't know it emulates correctly). I don't know what motivated to use BMP instead of the ffmpeg output? Maybe this problem?

Still loading the watchface with full screen picture takes 4sec.

drtonyr, 2023-11-13:
Do you have debug code, something that simply starts a timer, reads your file from SPINOR, writes it to the screen and then displays how long it took?

I'm interested in two things:

a) Where is this slowness coming from?   For example, I run P8 hardware, not Pinetime, getting your code out there may help find a factor.
b) The misalignment bug.  That's a strange one, I'd like to observe that so we can get it fixed.