wasp-os/wasp-os #485: Custom-built binaries don't load correctly

ajoliveau, 2024-01-20:
Hello everyone !
I cloned the project, and I'm trying to add a feature to wasp.py, but everything I build and push to the pinetime fails.
It works correctly in the simulator, the build completes without any error, I flash it on the pinetime using Gadgetbridge and when it boots it just displays "main.py" and nothing else.
If I run `waptool --console` I can connect to the watch. I can run `from shell import *` and I see that I have an empty logs folder, and a main.py file : 
```
# SPDX-License-Identifier: LGPL-3.0-or-later
# Copyright (C) 2020 Daniel Thompson

import wasp
from gadgetbridge import *
wasp.system.schedule()
```

A few strange things : 
- Even if I change nothing, everything I build and push has this error. If I just get the vanilla, v0.4 tag with no modifications, I have the same error
- Flashing the official release works, I have a working firmware
- If I modify the main.py file ( for instance I add a comment `# This is a test` ) the comment is not available in the main.py file after I flash the firmware
- I use Gadgetbridge because nRFConnect is not working (I downgraded to the 4.28.0 version, but when I try to flash it starts off reeaally quickly, then it's all over the place, before it reaches 100%... and stays there indefinitely). I tried the `ota-dfu` script in tools, but I get a `Exception at line 145: The command was not found or was not executable: gatttool.`. I'm on Arch, it looks like gattttool is deprecated ?

I'm not sure how to troubleshoot this error :/

drtonyr, 2024-01-21:
I don't have a pinetime, I'll try to help anyway.

You've got micropython working and (most of) the display driver because you see 'main.py' displayed.  You can also connect to the watch with wasptool.  So it seems like you are nearly there. 

> Even if I change nothing, everything I build and push has this error. If I just get the vanilla, v0.4 tag with no modifications, I have the same error
> Flashing the official release works, I have a working firmware

By 'official release' I assume you mean https://github.com/wasp-os/wasp-os/releases, so the problem is that the binary works but you can't build from source?  You are using Arch, so this could be a toolchain problem.  I've been hit by the wrong gcc before, we need v10 not v9 (`arm-none-eabi-gcc --version`)

> If I modify the main.py file ( for instance I add a comment # This is a test ) the comment is not available in the main.py file after I flash the firmware

I think the code only recreates it if it's not found.  If you modify on your host and push the modified version with wasptool it should stay modified.

> I use Gadgetbridge because nRFConnect is not working

It seems like you have something that works for you, so this isn't the problem. (`tools/ota-dfu/dfu.py -z build-p8/micropython.zip -a MA:CA:DD:ER:SS --legacy` works for me)

I hope others can help you further.

ajoliveau, 2024-01-22:
Thank you for these explanations !
It helps a lot.

I think I actually booted in safe mode everytime, so the instructions weren't run ?

Now when I flash a "normal" image (with code that I didn't change), it doesn't boot the first time but after I run the content of main.py manually it boots and reboots ok !

And when I flash my modifications and run through the main.py file, I can actually see the error :
```
>>> wasp.system.schedule()
Traceback (most recent call last):
  File "<stdin>", in <module>
AttributeError: 'module' object has no attribute 'system'
```

I'll try to debug my code, thank you so much.

closed 2024-01-22 by ajoliveau