wasp-os/wasp-os #493: Update micropython

fgaz, 2024-04-20:
The version of micropython we are using is a bit old and it's starting to get gcc warnings.

Initial attempt here https://github.com/wasp-os/micropython/tree/rebase-2024-04-05

GaryM0101, 2024-09-24:
@fgaz what do you need for this to go forward? I can test if you need help.

fgaz, 2024-09-24:
> @fgaz what do you need for this to go forward?

Time and motivation :smiling_face_with_tear: 

It's a lot of tedious work, and it's not like it's blocking anything else (for now)

jonnor, 2024-12-27:
I was just checking the MicroPython versions used in wasp-os. It seems that the MicroPython used in master is 1.16, and the branch above updates to 1.23. I am interested in running some machine learning code (using https://github.com/emlearn/emlearn-micropython/). However, this relies on native C modules, which got some critical fixes in 1.24

@fgaz what issues and open challenges do you face when updating MicroPython / what remains to be done? There seems to be a few wasp-os specific patches in the MicroPython fork, but nothing too crazy? Are there known incompatibilities or other issues that needs to be dealt with?

agittins, 2025-02-10:
> It's a lot of tedious work, and it's not like it's blocking anything else (for now)

Is moving to a newer micropython a part of unblocking some key bluetooth limitations, like not being able to send arbitrary adverts without disabling the nrf uart service? I am not clear on the details and whether it involves being able to use a different softdevice or using micropython's... new bt stack? Or perhaps it's one step in a long chain of dependencies, I'm not sure.

I check in on the project every few months in the hopes that I can one day bodge up a little python app to send BTHome adverts to use in home automation projects. I would very much like to be useful in the effort but it turns out I can write a *lot* of python code without it improving my C one iota! 😆

jonnor, 2025-04-25:
There might be a few Bluetooth fixes in recent MicroPython versions. However, the `nrf` port of MicroPython is not the most actively maintained, unfortunately. And as of MicroPython 1.25 there is no support for the new Bluetooth stack (though there is an MR for a while now). That issue is being tracked at https://github.com/micropython/micropython/issues/9851

closed 2026-11-14 by mbeijen