wasp-os/wasp-os #496: Install stalls at install of micropython on P8.

drjswalker, 2024-05-01:
Hi,
I am installing wasp-os on a Colmi P8 and the install went very well until it came time to use DaFlasher to install micropython.zip. I had finished installing the bootloader-daflasher.zip file and had the Wasp, start with arrow and bluetooth symbol in white in tthe bottom right hand corner.  I paused while I checked what came next and then started to install micropython.zip using daFlasher. It indicated the file had been selected and the file size (3000835Kb) and then "started DFU"  then "Connected" followed by the mac address of the watch. The "Starting Flashing" followed by Mac address followed by a please wait message syaing it takes up to 5 minutes. Nothing comes up on the progress bar, there is a zero data transfer rate number and nothing happens. I t sometimes disconnects and then reconnects going through the same cycle. The watch bluetooth symbol turns blue but nothing more happens. Any ideas?

tolidano, 2024-07-07:
I am also experiencing this. Which micro python.zip are you using? The one from the 0.4.1 release? I am about to try it with one manually built from latest master. There is another similar issue which may also have a working method, but I am on a Mac right now, and need to get my Linux machine running to use the ota-dfu tools: https://github.com/wasp-os/wasp-os/issues/222

fgaz, 2024-07-07:
Try the solution described in #475

fgaz, 2024-07-07:
alternatively wasptool should also work https://wasp-os.readthedocs.io/en/latest/install.html#wasptool-for-gnu-linux

tolidano, 2024-07-07:
https://github.com/wasp-os/wasp-os/issues/475#issuecomment-2212493506

tolidano, 2024-07-07:
I attempted to run wasp tool:

```
user@ea4532cb73da:~/wasp-os/tools/ota-dfu$ ./dfu.py -z ../../build-p8/micropython.zip -a F2:55:95:F3:01:28 --legacy
Exception at line 145: The command was not found or was not executable: gatttool.
```

I am in the container provided by `./tools/docker/shell` - I am on an M2 Mac, and it says this tool is missing. Should I run this outside the container? Where do I get gatttool? Is this only going to work on Linux?

tolidano, 2024-07-07:
trying wasptool on my linux machine (xubuntu on one notebook mix 4) and I get this:
```
root@1mix4:/home/tolidano/Dev/wasp-os/tools/ota-dfu# ./dfu.py -z ../../build-p8/micropython.zip -a F2:55:96:F3:01:28 --legacy
Sending file firmware.bin to F2:55:96:F3:01:28
Binary imge size: 311964
Binary CRC32: 1869247472
Connecting to F2:55:96:F3:01:28
Waiting for Image Size notification
Waiting for INIT DFU notification
Exception at line 166: a bytes-like object is required, not 'str'
root@1mix4:/home/tolidano/Dev/wasp-os/tools/ota-dfu# ./dfu.py -z ../../build-p8/micropython.zip -a F2:55:96:F3:01:28 --legacy
Sending file firmware.bin to F2:55:96:F3:01:28
Binary imge size: 311964
Binary CRC32: 1869247472
Connecting to F2:55:96:F3:01:28
Waiting for Image Size notification
Exception at line 166: Error in START_DFU procedure, reason: INVALID_STATE
root@1mix4:/home/tolidano/Dev/wasp-os/tools/ota-dfu# 
```

GaryM0101, 2024-07-07:
The best bet IMO is to get to Linux where you can use this or drop down
to the Nix shell. I gave up on docker long ago...

 

Links:
------
[1]
https://github.com/wasp-os/wasp-os/issues/496#issuecomment-2212498053
[2]
https://github.com/

tolidano, 2024-07-07:
So I made the change... 

```
root@1mix4:/home/tolidano/Dev/wasp-os/tools/ota-dfu# vim nrf_ble_dfu_controller.py
root@1mix4:/home/tolidano/Dev/wasp-os/tools/ota-dfu# ./dfu.py -z ../../build-p8/micropython.zip -a F2:55:96:F3:01:28 --legacy
Sending file firmware.bin to F2:55:96:F3:01:28
Binary imge size: 311964
Binary CRC32: 1869247472
Connecting to F2:55:96:F3:01:28
Waiting for Image Size notification
Waiting for INIT DFU notification
Exception at line 166: No notification received
```

And now no notification received.

tolidano, 2024-07-07:
So it appears to be a timing bug based on the thread there and there was a fix at some point...

tolidano, 2024-07-07:
https://github.com/wasp-os/ota-dfu-python/pull/3 this needs to be merged. or you can pull it by hand

tolidano, 2024-07-07:
so I added the origin `git remote add fix https://github.com/thiswillbeyourgithub/ota-dfu-python` 
then I `git pull bugfixes` 
then I removed the space before else on line 210
then ran the same command again and it is sending...slowly.

GaryM0101, 2024-07-07:
How far is the P8 away from the blue-tooth connection on the Linux box?
I would put it closest to the blue-tooth source so you don't timeout. 

The P8 blue-tooth can be pretty weak during install.

 

Links:
------
[1]
https://github.com/wasp-os/wasp-os/issues/496#issuecomment-2212520037
[2]
https://github.com/

tolidano, 2024-07-07:
So now it worked

```
root@1mix4:/home/tolidano/Dev/wasp-os/tools/ota-dfu# ./dfu.py -z ../../build-p8/micropython.zip -a F2:55:96:F3:01:28 --legacy
Sending file firmware.bin to F2:55:96:F3:01:28
Binary imge size: 311964
Binary CRC32: 1869247472
Connecting to F2:55:96:F3:01:28
Waiting for Image Size notification
Waiting for INIT DFU notification
Begin DFU
[##################################################] 100.0%  (304 of 304 kb)

Upload complete in 7 minutes and 7 seconds
Waiting for DFU complete notification
Waiting for Firmware Validation notification
Activate and reset
```

But it still just shows the waspos and the start button and the little bluetooth, no interface with the clock

tolidano, 2024-07-07:
When I hit the button to "start" it shows JUST the wasp, then the screen goes black, then it flashes back to the wasp with the start and the bluetooth icon

GaryM0101, 2024-07-07:
boot up in safe mode and see what the main.py is reading using wasptool
console. You might have to reload the main.py file. 

https://wasp-os.readthedocs.io/en/latest/install.html#id35 

 

Links:
------
[1]
https://github.com/wasp-os/wasp-os/issues/496#issuecomment-2212536281
[2]
https://github.com/

tolidano, 2024-07-07:
Cannot get to safe mode by holding down the physical button. 

Also thank you for your help so far. I feel we are close!

tolidano, 2024-07-08:
Got this working by #222 and #343