wasp-os/wasp-os #497: make submodules no longer works - "Support for password authentication was removed on August 13, 2021."

tomchiverton, 2024-06-26:
```
...
Submodule path 'lib/nrfx': checked out '3d64bcae3de719b4f67ed0fda7d1ca774de1c105'
(cd wasp/modules/bma42x-upy; git submodule update --init)
Submodule 'BMA423-Sensor-API' (https://github.com/BoschSensortec/BMA423-Sensor-API) registered for path 'BMA423-Sensor-API'
Cloning into '/home/falken/workspace/wasp-os/wasp/modules/bma42x-upy/BMA423-Sensor-API'...
remote: Invalid username or password.
fatal: Authentication failed for 'https://github.com/BoschSensortec/BMA423-Sensor-API/'
fatal: clone of 'https://github.com/BoschSensortec/BMA423-Sensor-API' into submodule path '/home/falken/workspace/wasp-os/wasp/modules/bma42x-upy/BMA423-Sensor-API' failed
Failed to clone 'BMA423-Sensor-API'. Retry scheduled
Cloning into '/home/falken/workspace/wasp-os/wasp/modules/bma42x-upy/BMA423-Sensor-API'...
remote: Support for password authentication was removed on August 13, 2021.
remote: Please see https://docs.github.com/get-started/getting-started-with-git/about-remote-repositories#cloning-with-https-urls for information on currently recommended modes of authentication.
fatal: Authentication failed for 'https://github.com/BoschSensortec/BMA423-Sensor-API/'
fatal: clone of 'https://github.com/BoschSensortec/BMA423-Sensor-API' into submodule path '/home/falken/workspace/wasp-os/wasp/modules/bma42x-upy/BMA423-Sensor-API' failed
Failed to clone 'BMA423-Sensor-API' a second time, aborting
make: *** [Makefile:37: submodules] Error 1

```

Ubuntu 24.04 LTS

tomchiverton, 2024-06-26:
I created a token, and would update the waspos docs with links to https://docs.github.com/en/authentication/keeping-your-account-and-data-secure/managing-your-personal-access-tokens#creating-a-fine-grained-personal-access-token

But it seems the repo is not found ?

```
...
fatal: clone of 'https://github.com/BoschSensortec/BMA423-Sensor-API' into submodule path '/home/falken/workspace/wasp-os/wasp/modules/bma42x-upy/BMA423-Sensor-API' failed
Failed to clone 'BMA423-Sensor-API'. Retry scheduled
Cloning into '/home/falken/workspace/wasp-os/wasp/modules/bma42x-upy/BMA423-Sensor-API'...
remote: Repository not found.
fatal: repository 'https://github.com/BoschSensortec/BMA423-Sensor-API/' not found
fatal: clone of 'https://github.com/BoschSensortec/BMA423-Sensor-API' into submodule path '/home/falken/workspace/wasp-os/wasp/modules/bma42x-upy/BMA423-Sensor-API' failed
Failed to clone 'BMA423-Sensor-API' a second time, aborting
make: *** [Makefile:37: submodules] Error 1

```

tomchiverton, 2024-06-26:
https://github.com/BoschSensortec no longer lists a BMA423-Sensor-API project :/

tomchiverton, 2024-06-26:
https://github.com/Deadroad/BMA423-Sensor-API might be a still visible fork but I don't know enough about Git submodules to confidently suggest where to change this. Maybe this project needs to even create it's own fork ?

tomchiverton, 2024-06-26:
Also the included mpy-cross fails to compile, due to these, which might be GCC13 related ?
```
falken@frame:~/workspace/wasp-os/micropython/mpy-cross$ git diff
diff --git a/mpy-cross/main.c b/mpy-cross/main.c
index 635e53a71..eba329027 100644
--- a/mpy-cross/main.c
+++ b/mpy-cross/main.c
@@ -339,7 +339,7 @@ int main(int argc, char **argv) {
     return main_(argc, argv);
 }
 
-uint mp_import_stat(const char *path) {
+mp_import_stat_t mp_import_stat(const char *path) {
     (void)path;
     return MP_IMPORT_STAT_NO_EXIST;
 }
diff --git a/py/stackctrl.c b/py/stackctrl.c
index c2f3adb5e..6ba468ff7 100644
--- a/py/stackctrl.c
+++ b/py/stackctrl.c
@@ -28,12 +28,19 @@
 #include "py/stackctrl.h"
 
 void mp_stack_ctrl_init(void) {
+    #if __GNUC__ >= 13
+    #pragma GCC diagnostic push
+    #pragma GCC diagnostic ignored "-Wdangling-pointer"
+    #endif
     volatile int stack_dummy;
     MP_STATE_THREAD(stack_top) = (char *)&stack_dummy;
 }
 
 void mp_stack_set_top(void *top) {
     MP_STATE_THREAD(stack_top) = top;
+    #if __GNUC__ >= 13
+    #pragma GCC diagnostic pop
+    #endif
 }
 
 mp_uint_t mp_stack_usage(void) {
```

fgaz, 2024-06-28:
Oof. Looks like we'll have to mirror BMA423-Sensor-API.

> Also the included mpy-cross fails to compile, due to these, which might be GCC13 related ?

Yes, use GCC 11 if possible. That's what we use in the docker image and nix shell. #494 fixes it for GCC 13.

fgaz, 2024-06-28:
I created https://github.com/wasp-os/BMA423-Sensor-API, now I'll update the submodules.

PR #498 (2024-06-28): bma42x-upy: update submodule
The remote of its submodule disappeared. Update it to a fork under the wasp-os organization.

Fixes #497
closed 2024-06-28 by daniel-thompson