wasp-os/wasp-os #501: Smartwatch - MCU JQ7011

token47, 2024-08-05:
I have a smartwatch that has this MCU "JQ7011", and I cannot find anything on the internet about it (except many watch models saying they have this CPU).

Does anyone have any information on them? Are they compatible (even with some extra efforts) with wasp-os?

References about the watch:
https://xdaforums.com/t/lj736-d-ultra-custom-watchface-procedure.4655069/
https://www.muzzaonline.com/product-page/watch-9-ultra-smart-watch-49mm-2023-new-nfc-men-women-gps-track-bluetooth-call
https://www.valdus.com/product/i15-ultra-max-smart-watch.html

token47, 2024-08-05:
[image]

[image]

drtonyr, 2024-08-05:
The documentation says https://wasp-os.readthedocs.io/en/latest/install.html#device-support

As the CPU is listed as JQ7011 not nRF52832 I'd have thought you were a long long way away from booting.

Rumidom, 2025-05-07:
not sure if this helps in any way but the new COLMI P81 uses a JieLi Tech JL7012F6 it seems like manufacturers are opting for more integrated chips

datasheet:
https://jlcpcb.com/api/file/downloadByFileSystemAccessId/8590906238253551616

greg1985, 2025-07-08:
Hi there. I have the same model. Anybody knows the driver IC of the display and the pinout? I know the touch is read by external IC, but I just need to connect the display to an esp32. 
Thanks in advance.

X-TechPro, 2026-02-28:
> Hi there. I have the same model. Anybody knows the driver IC of the display and the pinout? I know the touch is read by external IC, but I just need to connect the display to an esp32. Thanks in advance.

Same. I also want to use this screen. Any updates?