wasp-os/wasp-os #505: GadgetBridge says PineDFU is unsupported and DaFlsher connects but doesn't transfer OS install data

arghunter, 2024-09-24:
Exactly as the title states. The install isn't working

GaryM0101, 2024-09-24:
See the docs to get the workaround... 

https://wasp-os.readthedocs.io/en/latest/install.html#main-os-gadgetbridge

Cheers! 

 

Links:
------
[1] https://github.com/wasp-os/wasp-os/issues/505
[2]
https://github.com/

arghunter, 2024-09-24:
Thank you so much,  and how can i swap between wasp and infinitime?

GaryM0101, 2024-09-24:
See the docs for restoring the Pintime factory firmware... 

https://wasp-os.readthedocs.io/en/latest/install.html#gadgetbridge-for-android

And please close this issue. 

Cheers!

 

Links:
------
[1]
https://github.com/wasp-os/wasp-os/issues/505#issuecomment-2369987512
[2]
https://github.com/

closed 2024-09-24 by arghunter