wasp-os/wasp-os #519: Pinetime: wasp-reloader needs to use new Infinitime bootloader 1.0.1 for builing factory image

GaryM0101, 2024-12-06:
A new [bootloader for Infinitime](https://github.com/InfiniTimeOrg/pinetime-mcuboot-bootloader/releases/tag/1.0.1) was patched for the new SPI flash memory chip that was put in for the new batch of Pinetime devices that started in September.

We saw a recent issue where a user tried to flash the old bootloader 1.0.0 onto a new Pinetime with the new SPI flash memory chip. They soft-bricked the device.  

The wasp-reloader needs to use the new Infinitime bootloader 1.0.1 for building factory image (reloader-factory.zip)

Folks wanting to go back to Infinitime with the new Pinetime SPI chip will undoubtedly soft-brick their device in doing so.

daniel-thompson, 2024-12-26:
I have to admit I have almost no recollection of how any of this stuff works but I went through the wasp-reloader build system until I figured things out.

Instructions on how to update the Infinitime bootloader are here:
https://github.com/wasp-os/wasp-reloader/blob/14f85fc92180c0b29f36400480b54a571dc510b8/src/boards/pinetime/factory_image.h

GaryM0101, 2025-01-12:
I built a test version of the factory-reloader.zip with the new 1.0.1 bootloader and successfully deployed it on an older devkit. The reloader still uses the pinetime-mcuboot-recovery-image-0.11.0. We'll still need to test it for the new SPI chip and formally get a PR going to deploy it.

Not sure if any other changes will be needed for WaspOS but we'll find out here shortly. For now I'll attach the updated file here for folks that might need it in case they didn't see the warnings.

[reloader-factory.zip](https://github.com/user-attachments/files/18390367/reloader-factory.zip)

GaryM0101, 2025-02-01:
The new Wasp "reloader" tool that flashes the InfiniTime bootloader v 1.0.1 was successfully tested on the pinetime with the new BY25Q32 memory chip the TL Lim shipped to me last week. 

We'll get a PR going to get the new factory_image.h into the release but folks can now use the [reloader-factory.zip](https://github.com/user-attachments/files/18390367/reloader-factory.zip) to revert back to the pinetime-mcuboot-recovery-image and flash the latest Infinitime firmware.

From the Wasp side I had no issues flashing the reloader-mcuboot and the micropython v0.4 firmware. I was able to access the SPI flash area without any issues.

closed 2025-02-02 by daniel-thompson

pawel-ch, 2025-12-24:
So should we still refrain from switching OSes as per https://wiki.pine64.org/wiki/Switching_your_PineTime_between_InfiniTime_and_Wasp-os and https://pine64.org/2024/12/29/please-refrain-from-installing-waspos-on-new-pinetimes-for-the-moment/? Or maybe not updating the documentation and not publishing a new blog post about this was an oversight?

GaryM0101, 2026-02-21:
You got lucky. As long as you use the reloader-factory.zip in the above message to revert back to Infinitime, there should be no problem.