wasp-os/wasp-os #521: BangleJS Support

bobrippling, 2025-01-16:
I'm interested in seeing how this runs on the [BangleJS2](https://www.espruino.com/Bangle.js2). I was going to try a [custom build](https://wasp-os.readthedocs.io/en/latest/install.html#custom-builds), but was wondering, are there any references based on porting to a new architecture?

jonnor, 2025-04-25:
BangleJS2 is nRF52840, which is very similar to nRF52832. Same microcontroller family and MCU architecture. But everything else is likely to be different. So first order of business will just be to get MicroPython running. Probably a stock MicroPython build will be easiest. And then you need to add drivers for every single peripheral. Once you have all of that inplace, one can consider trying to get a particular codebase like Wasp OS to run.

mbeijen, 2026-11-14:
I don't have the hardware to work on this; it seems feasible to add Bangle as supported platform but it would require a dedicated person contributing to and maintaining this platform.